elFarto/nvidia-vaapi-driver #362: libva error: /usr/lib/dri/nvidia_drv_video.so init failed

jeevithakannan2, 2025-04-05:
Before closing the issue as a duplicate please read through the context.

I don't know if this is the correct place to ask this I am using MUX switch to turn of my integrated GPU and run only on NVIDIA.

I have installed nvidia-vaapi-driver and all the related dependencies for running nvidia gpu.

When try to open electron applications I encounter the `libva error: /usr/lib/dri/nvidia_drv_video.so init failed` error. After going through many issues in this repo I coudn't find the solution. But setting NVD_GPU=0 resolves the issue.

My GPU ID for NVIDIA is 0 because I am using MUX and the intel iGPU's ID is 1

I cannot understand why settings NVD_GPU=0 resolves the error while unsetting it results in an error

thesword53, 2025-04-07:
Hello @jeevithakannan2,

Did you run Firefox with the Nvidia GPU ? You can check with `about:support` URL.. Try unsetting `LIBVA_DRIVER_NAME` if you have it set to `nvidia`.

jeevithakannan2, 2025-04-08:
> Hello [@jeevithakannan2](https://github.com/jeevithakannan2),
> 
> Did you run Firefox with the Nvidia GPU ? You can check with `about:support` URL.. Try unsetting `LIBVA_DRIVER_NAME` if you have it set to `nvidia`.

I am running firefox with NVidia GPU but the hw dec is not working. Unsetting LIBVA_DRIVER_NAME does resolve the issue.

Is hw accel broken on firefox?

thesword53, 2025-04-10:
Firefox 137 changed the config flag to force enable VAAPI from `media.ffmpeg.vaapi.enabled` to `media.hardware-video-decoding.force-enabled`.

> When try to open electron applications I encounter the libva error: /usr/lib/dri/nvidia_drv_video.so init failed error. After going through many issues related to libva errors I coudn't find the solution. But setting NVD_GPU=0 resolves the issue.

To get VAAPI working on Electron/Chromium  you can check here: https://github.com/elFarto/nvidia-vaapi-driver/issues/5

> Doesn't NVD_GPU automatically detect the NVIDIA GPU ?

`NVD_GPU` chooses between Nvidia GPUs when you have multiple Nvidia GPUs. It will never choose your Intel GPU.

jeevithakannan2, 2025-05-26:
> Firefox 137 changed the config flag to force enable VAAPI from `media.ffmpeg.vaapi.enabled` to `media.hardware-video-decoding.force-enabled`.

Yes this fixes the issue with firefox
 
> > Doesn't NVD_GPU automatically detect the NVIDIA GPU ?
> 
> `NVD_GPU` chooses between Nvidia GPUs when you have multiple Nvidia GPUs. It will never choose your Intel GPU.

Unsetting LIBVA_DRIVER_NAME=nvidia resolves the issue

closed 2026-09-26 by elFarto