elFarto/nvidia-vaapi-driver #363: Video Engine Utilization 0% Debian 12 Firefox-esr

sigod12, 2025-04-05:
Hi, I followed these steps on the elFarto/nvidia-vaapi-driver website on Debian 12 and it still didn't work:

sudo apt install meson gstreamer1.0-plugins-bad libffmpeg-nvenc-dev libva-dev libegl-dev libgstreamer-plugins-bad1.0-dev

git clone https://github.com/elFarto/nvidia-vaapi-driver
cd nvidia-vaapi-driver
meson setup build
meson install -C build

Which installed the driver

# echo "options nvidia-drm modeset=1" >> /etc/modprobe.d/nvidia-options.conf

Checked to see modeset was working after a restart and it was.

I set three env variables in /etc/environment and restarted
LIBVA_DRIVER_NAME=nvidia
MOZ_DISABLE_RDD_SANDBOX=1
__EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/10_nvidia.json

Enabled all the config options in firefox-esr 128.9.0 

Set LIBVA_MESSAGING_LEVEL=1 in etc/libva.conf 

I tried to play a video with MPV: mpv --hwdec=vaapi /mnt/Media2TB/2TB/Movies/[2012]BRRip XviD-ETRG.avi
[file] This is a directory - adding to playlist.

Playing: [2012]BRRip XviD-ETRG.avi
 (+) Video --vid=1 (mpeg4 640x352 24.000fps)
 (+) Audio --aid=1 (mp3 2ch 48000Hz)
AO: [pipewire] 48000Hz stereo 2ch floatp
VO: [gpu] 640x352 yuv420p
(Paused) AV: 00:31:50 / 01:35:44 (33%) A-V:  0.000 Cache: 982s/150MB

MPV didn't show any video engine utilization in nvidia settings and neither did firefox. I just read that stable builds of firefox don't allow hardware video acceleration is this true? Do I need to use firefox nightly?

elFarto, 2025-04-19:
Can you post the output of vainfo?

sigod12, 2025-05-15:
sigod@debian:~$ vainfo
vainfo: VA-API version: 1.17 (libva 2.12.0)
vainfo: Driver version: VA-API NVDEC driver [direct backend]
vainfo: Supported profile and entrypoints
      VAProfileMPEG2Simple            :	VAEntrypointVLD
      VAProfileMPEG2Main              :	VAEntrypointVLD
      VAProfileVC1Simple              :	VAEntrypointVLD
      VAProfileVC1Main                :	VAEntrypointVLD
      VAProfileVC1Advanced            :	VAEntrypointVLD
      VAProfileH264Main               :	VAEntrypointVLD
      VAProfileH264High               :	VAEntrypointVLD
      VAProfileH264ConstrainedBaseline:	VAEntrypointVLD

closed 2026-09-26 by elFarto