elFarto/nvidia-vaapi-driver #364: vainfo segmentation fault

quantumfrost, 2025-04-10:
I had things running fine, but now I'm suddenly getting a segmentation fault when running `vainfo` or anything with vaapi like `mpv --hwdec=vaapi`

```
$ NVD_LOG=1 NVD_BACKEND=direct vainfo
libva info: VA-API version 1.20.0
Segmentation fault (core dumped)
```

dmesg:
```
vainfo[40147]: segfault at 0 ip 000072562da58f1f sp 00007ffdf658e490 error 4 in libva-wayland.so.2.2000.0[2f1f,72562da58000+2000] likely on CPU 1 (core 1, socket 0)
Code: 48 89 e5 41 57 41 56 49 89 f6 41 55 49 89 d5 41 54 53 48 81 ec 98 01 00 00 64 48 8b 04 25 28 00 00 00 48 89 45 c8 48 8b 47 68 <8b> 38 e8 8a f4 ff ff 48 85 c0 0f 84 01 01 00 00 48 8b 78 10 48 89
```
trace:

```
libva info: VA-API version 1.20.0
==41924== Invalid read of size 4
==41924==    at 0x49C0F1F: ??? (in /usr/lib/x86_64-linux-gnu/libva-wayland.so.2.2000.0)
==41924==    by 0x49C929D: vaInitialize (in /usr/lib/x86_64-linux-gnu/libva.so.2.2000.0)
==41924==    by 0x10A8DF: ??? (in /usr/bin/vainfo)
==41924==    by 0x4A2F1C9: (below main) (libc_start_call_main.h:58)
==41924==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==41924==
==41924==
==41924== Process terminating with default action of signal 11 (SIGSEGV)
==41924==  Access not within mapped region at address 0x0
==41924==    at 0x49C0F1F: ??? (in /usr/lib/x86_64-linux-gnu/libva-wayland.so.2.2000.0)
==41924==    by 0x49C929D: vaInitialize (in /usr/lib/x86_64-linux-gnu/libva.so.2.2000.0)
==41924==    by 0x10A8DF: ??? (in /usr/bin/vainfo)
==41924==    by 0x4A2F1C9: (below main) (libc_start_call_main.h:58)
==41924==  If you believe this happened as a result of a stack
==41924==  overflow in your program's main thread (unlikely but
==41924==  possible), you can try to increase the size of the
==41924==  main thread stack using the --main-stacksize= flag.
==41924==  The main thread stack size used in this run was 8388608.
==41924==
==41924== HEAP SUMMARY:
==41924==     in use at exit: 18,216 bytes in 13 blocks
==41924==   total heap usage: 93 allocs, 80 frees, 40,068 bytes allocated
==41924==
==41924== Searching for pointers to 13 not-freed blocks
==41924== Checked 263,624 bytes
==41924==
==41924== LEAK SUMMARY:
==41924==    definitely lost: 0 bytes in 0 blocks
==41924==    indirectly lost: 0 bytes in 0 blocks
==41924==      possibly lost: 0 bytes in 0 blocks
==41924==    still reachable: 18,216 bytes in 13 blocks
==41924==         suppressed: 0 bytes in 0 blocks
==41924== Rerun with --leak-check=full to see details of leaked memory
==41924==
==41924== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
==41924==
==41924== 1 errors in context 1 of 1:
==41924== Invalid read of size 4
==41924==    at 0x49C0F1F: ??? (in /usr/lib/x86_64-linux-gnu/libva-wayland.so.2.2000.0)
==41924==    by 0x49C929D: vaInitialize (in /usr/lib/x86_64-linux-gnu/libva.so.2.2000.0)
==41924==    by 0x10A8DF: ??? (in /usr/bin/vainfo)
==41924==    by 0x4A2F1C9: (below main) (libc_start_call_main.h:58)
==41924==  Address 0x0 is not stack'd, malloc'd or (recently) free'd
==41924==
==41924== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
Segmentation fault (core dumped)
```
Thanks for your help

elFarto, 2025-04-19:
libva 1.20 is a bit old at this point. Is there a new version of that package available for your distro?

quantumfrost, 2025-04-20:
I have libva 2.20 installed (Ubuntu 24.04). Is VA-API version something different, or is there an issue with loading the correct library?

quantumfrost, 2025-06-11:
Hi, I would appreciate it if someone could help me figure this out, or point me to other resources/community/discord etc.

quantumfrost, 2025-06-17:
I don't know why, but installing the same version of the driver from `ppa:graphics-drivers/ppa` fixed the issue. I had previously purged and reinstalled from the official repo many times and it hadn't helped.

closed 2025-06-17 by quantumfrost