elFarto/nvidia-vaapi-driver #365: [8k video] NS_ERROR_DOM_MEDIA_FATAL_ERR (0x806e0005): file /usr/src/debug/firefox-developer-edition/firefox-138.0/dom/media/MediaDecoderStateMachineBase.cpp:168

Doaxan, 2025-04-11:
Actual problem:
1) Open https://www.youtube.com/watch?v=zCLOJ9j1k2Y
2) Set 8k 4320p60
3) After 3 seconds video stopped, quality auto changed to 1080p60. 
4) Error in logs Decode error: `NS_ERROR_DOM_MEDIA_FATAL_ERR (0x806e0005)`. Please see 
[doaxan_nvidia_log_from_github.txt](https://github.com/user-attachments/files/19712962/doaxan_nvidia_log_from_github.txt)

Old issue, before upgrading to master:
```
     28874.766999480 [79805-80930] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c: 955        nvCreateSurfaces2 Creating surface 1920x1088, format 1 (0x7ca3b0e462e0)
     28874.767411915 [79805-80930] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c:1342             nvEndPicture cuvidDecodePicture failed: 1
[Child 80621, MediaDecoderStateMachine #1] WARNING: Decoder=7765278a1500 Decode error: NS_ERROR_DOM_MEDIA_FATAL_ERR (0x806e0005): file /usr/src/debug/firefox-developer-edition/firefox-138.0/dom/media/MediaDecoderStateMachineBase.cpp:168
     28874.768169788 [79805-80930] ../nvidia-vaapi-driver-0.0.13/src/vabackend.c: 989        nvDestroySurfaces Destroying surface 6 (0x7ca3b0ec9060)
```
I just want to tell those who experienced the same error as me when scrolling through the video on the twitch, update nvidia-vaapi-driver to 0.0.13.r17.gc519e97 master version at the moment. For those who are arch-based, just use libva-nvidia-driver-git from aur. Also add the necessary variables to /etc/environment, without them I did not work hardware decoding on nvidia 570.133.07. 

Many thanks to the developers for the work done!

Shoun2137, 2025-04-11:
Can confirm, I have the same problems, same stuff in logs, can't play any YT stream or any twitter video (like, at all). Moreover FF based browsers since 137 (with clean profiles) entirely lost the ability to play videos in software mode, so even when hwaccel is available, it just bails on any video/audio media that's on RED here: https://i.imgur.com/ir5LTNl.png. GTX 1060, 570.133.07, ArtixLinux, ffmpeg, libva, various gst plugins and whatnot was freshly provided from distro's downstream and then updated to git upstream - and it still doesn't matter, cause it doesn't work.

thesword53, 2025-04-11:
I created the pull request https://github.com/elFarto/nvidia-vaapi-driver/pull/366. Maybe it can help.

Doaxan, 2025-04-12:
@thesword53 unfortunately:
```
doaxan@b660m ~> MOZ_DISABLE_RDD_SANDBOX=0 NVD_LOG=1 MOZ_GFX_DEBUG=1 firefox-developer-edition
GLX_TEST: childgltest start
GLX_TEST: get_pci_status start
GLX_TEST: get_pci_status finished
GLX_TEST: wayland_egltest start
GLX_TEST: get_egl_status start
GLX_TEST: get_egl_gl_status start
GLX_TEST: get_egl_gl_status finished
GLX_TEST: get_egl_status finished with return: 1
GLX_TEST: wayland_egltest finished
PCI_VENDOR_ID
0x10de
PCI_DEVICE_ID
0x2204
VENDOR
NVIDIA Corporation
RENDERER
NVIDIA GeForce RTX 3090/PCIe/SSE2
VERSION
4.6.0 NVIDIA 570.133.07
TFP
TRUE
DRM_RENDERDEVICE
/dev/dri/renderD128
MESA_VENDOR_ID
0x10de
MESA_DEVICE_ID
0x2204
TEST_TYPE
EGL
GLX_TEST: childgltest finished
vaapitest start, device /dev/dri/renderD128
vaInitialize finished
Profile: H264Main
Profile: H264High
Profile: H264ConstrainedBaseline
Profile: HEVCMain
Profile: VP8Version0_3
Profile: VP9Profile0
Profile: AV1Profile0
Profile: HEVCMain10
Profile: HEVCMain12
Profile: VP9Profile2
vaapitest finished
VAAPI_SUPPORTED
TRUE
VAAPI_HWCODECS
496
      9998.897835723 [87540-87573] ../src/vabackend.c:2290       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver
      9998.897845227 [87540-87573] ../src/vabackend.c:2298       __vaDriverInit_1_0 Got DRM FD: 1 23
      9998.897850820 [87540-87573] ../src/vabackend.c:2310       __vaDriverInit_1_0 Now have 0 (0 max) instances
      9998.897857849 [87540-87573] ../src/vabackend.c:2337       __vaDriverInit_1_0 Selecting Direct backend
      9998.905304848 [87540-87573] ../src/direct/nv-driver.c: 292            init_nvdriver Initing nvdriver...
      9998.905334262 [87540-87573] ../src/direct/nv-driver.c: 310            init_nvdriver NVIDIA kernel driver version: 570.133.07, major version: 570, minor version: 133
      9998.905338098 [87540-87573] ../src/direct/nv-driver.c: 317            init_nvdriver Got dev info: 100 1 2 6
      9999.004371717 [87540-87573] ../src/vabackend.c:1546      nvQueryImageFormats In nvQueryImageFormats
      9999.200648942 [87540-87573] ../src/vabackend.c: 726           nvCreateConfig got profile: 6 with 0 attributes
      9999.200668528 [87540-87573] ../src/vabackend.c:1902 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      9999.200672697 [87540-87573] ../src/vabackend.c:1902 nvQuerySurfaceAttributes with 4 (8) 0x72b1cf18f580 8
      9999.210074950 [87540-87573] ../src/vabackend.c:1969 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      9999.210100975 [87540-87573] ../src/vabackend.c: 726           nvCreateConfig got profile: 7 with 0 attributes
      9999.210108546 [87540-87573] ../src/vabackend.c:1902 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      9999.210110920 [87540-87573] ../src/vabackend.c:1902 nvQuerySurfaceAttributes with 4 (8) 0x72b1cf18f580 8
      9999.219630672 [87540-87573] ../src/vabackend.c:1969 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      9999.219655209 [87540-87573] ../src/vabackend.c: 726           nvCreateConfig got profile: 13 with 0 attributes
      9999.219666371 [87540-87573] ../src/vabackend.c:1902 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      9999.219670349 [87540-87573] ../src/vabackend.c:1902 nvQuerySurfaceAttributes with 4 (8) 0x72b1cf18f580 8
      9999.229390053 [87540-87573] ../src/vabackend.c:1969 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
[... 381 lines cut ...]
     10016.614921576 [87540-88733] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x72b1a5bcdd90 1920x1080
     10016.727876415 [87540-88733] ../src/vabackend.c:1023        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x72b1763917a0)
     10016.728156862 [87540-88733] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x72b1a81df050 1920x1080
     10019.138506682 [87540-88733] ../src/vabackend.c:1209         nvDestroyContext Destroying context: 4
     10019.138517771 [87540-88733] ../src/vabackend.c: 321           destroyContext Signaling resolve thread to exit
     10019.138520667 [87540-88733] ../src/vabackend.c: 327           destroyContext Waiting for resolve thread to exit
     10019.138541722 [87540-88736] ../src/vabackend.c: 466          resolveSurfaces [RT] Resolve thread for 0x72b1a5b74000 exiting
     10019.138570318 [87540-88733] ../src/vabackend.c: 329           destroyContext pthread_timedjoin_np finished with 0
     10019.243673203 [87540-88733] ../src/vabackend.c:1057        nvDestroySurfaces Destroying surface 3 (0x72b1a5b8f920)
     10019.244092470 [87540-88733] ../src/vabackend.c:1057        nvDestroySurfaces Destroying surface 12 (0x72b1763917a0)
     10019.244306995 [87540-88733] ../src/vabackend.c:1057        nvDestroySurfaces Destroying surface 9 (0x72b1a81ddde0)
     10019.244623618 [87540-88733] ../src/vabackend.c:1057        nvDestroySurfaces Destroying surface 8 (0x72b1a80b06a0)
     10019.244828142 [87540-88733] ../src/vabackend.c:1057        nvDestroySurfaces Destroying surface 10 (0x72b1a81dee20)
     10019.245031635 [87540-88733] ../src/vabackend.c:1057        nvDestroySurfaces Destroying surface 2 (0x72b1a5b8e5c0)
     10019.245226848 [87540-88733] ../src/vabackend.c:1057        nvDestroySurfaces Destroying surface 0 (0x72b1a5a17a60)
     10019.245430496 [87540-88733] ../src/vabackend.c:1057        nvDestroySurfaces Destroying surface 1 (0x72b1a5b8ec00)
[... cut ...]
     10019.245981803 [87540-88733] ../src/vabackend.c:1057        nvDestroySurfaces Destroying surface 5 (0x72b1a81d70c0)
     10019.246166533 [87540-88733] ../src/vabackend.c:1057        nvDestroySurfaces Destroying surface 4 (0x72b1a81d8240)
     10019.246343542 [87540-88733] ../src/vabackend.c:1057        nvDestroySurfaces Destroying surface 11 (0x72b1763910c0)
     10019.246540047 [87540-88733] ../src/vabackend.c:2198              nvTerminate Terminating 0x72b1cf15e120
     10019.247366446 [87540-88733] ../src/vabackend.c:2212              nvTerminate Now have 0 (0 max) instances
doaxan@b660m ~> 
```

thesword53, 2025-05-03:
> [@thesword53](https://github.com/thesword53) unfortunately:

I modified https://github.com/elFarto/nvidia-vaapi-driver/pull/366 can you check again if you still have the issue ?

ManuLinares, 2025-05-03:
> I modified [#366](https://github.com/elFarto/nvidia-vaapi-driver/pull/366) can you check again if you still have the issue ?

Same issue

<details>
<summary>Here is the log, same steps as OP.</summary>

```
~ $ env NVD_LOG=1 firefox
    138512.067527744 [1210569-1210604] ../src/vabackend.c:2240       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver
    138512.067539198 [1210569-1210604] ../src/vabackend.c:2248       __vaDriverInit_1_0 Got DRM FD: 1 23
    138512.067542993 [1210569-1210604] ../src/vabackend.c:2260       __vaDriverInit_1_0 Now have 0 (0 max) instances
    138512.067548556 [1210569-1210604] ../src/vabackend.c:2287       __vaDriverInit_1_0 Selecting Direct backend
    138512.075063717 [1210569-1210604] ../src/direct/nv-driver.c: 305            init_nvdriver Initing nvdriver...
    138512.075091919 [1210569-1210604] ../src/direct/nv-driver.c: 323            init_nvdriver NVIDIA kernel driver version: 570.144, major version: 570, minor version: 144
    138512.075095322 [1210569-1210604] ../src/direct/nv-driver.c: 330            init_nvdriver Got dev info: 100 1 2 6
    138512.158108252 [1210569-1210604] ../src/vabackend.c:1486      nvQueryImageFormats In nvQueryImageFormats
    138512.276088743 [1210569-1210604] ../src/vabackend.c: 674           nvCreateConfig got profile: 6 with 0 attributes
    138512.276101629 [1210569-1210604] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 4 (8) (nil) 0
    138512.276104616 [1210569-1210604] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 4 (8) 0x79c6cf116340 8
    138512.282836057 [1210569-1210604] ../src/vabackend.c:1919 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
    138512.282853057 [1210569-1210604] ../src/vabackend.c: 674           nvCreateConfig got profile: 7 with 0 attributes
    138512.282857868 [1210569-1210604] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 4 (8) (nil) 0
    138512.282859309 [1210569-1210604] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 4 (8) 0x79c6cf116280 8
    138512.289437500 [1210569-1210604] ../src/vabackend.c:1919 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
    138512.289450458 [1210569-1210604] ../src/vabackend.c: 674           nvCreateConfig got profile: 13 with 0 attributes
    138512.289454446 [1210569-1210604] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 4 (8) (nil) 0
    138512.289455906 [1210569-1210604] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 4 (8) 0x79c6cf116e80 8
    138512.296240534 [1210569-1210604] ../src/vabackend.c:1919 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
    138512.296256274 [1210569-1210604] ../src/vabackend.c: 674           nvCreateConfig got profile: 17 with 0 attributes
    138512.296259945 [1210569-1210604] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 8 (8) (nil) 0
    138512.296261465 [1210569-1210604] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 8 (8) 0x79c6cf116b80 8
    138512.302664074 [1210569-1210604] ../src/vabackend.c:1919 nvQuerySurfaceAttributes Returning constraints: width: 144 - 8192, height: 144 - 8192
    138512.302677718 [1210569-1210604] ../src/vabackend.c: 674           nvCreateConfig got profile: 14 with 0 attributes
    138512.302680974 [1210569-1210604] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 9 (8) (nil) 0
    138512.302682412 [1210569-1210604] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 9 (8) 0x79c6cf116880 8
    138512.308916628 [1210569-1210604] ../src/vabackend.c:1919 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
    138512.308941116 [1210569-1210604] ../src/vabackend.c: 674           nvCreateConfig got profile: 19 with 0 attributes
    138512.308946478 [1210569-1210604] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 10 (8) (nil) 0
    138512.308947947 [1210569-1210604] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 10 (8) 0x79c6cf116880 8
    138512.315508906 [1210569-1210604] ../src/vabackend.c:1919 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
    138512.315526734 [1210569-1210604] ../src/vabackend.c: 674           nvCreateConfig got profile: 32 with 0 attributes
    138512.315530021 [1210569-1210604] ../src/vabackend.c: 737           nvCreateConfig Unable to determine surface type for VP9/AV1 codec due to no RTFormat specified.
    138512.315534452 [1210569-1210604] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 11 (8) (nil) 0
    138512.315536329 [1210569-1210604] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 11 (8) 0x79c6cf116280 8
    138512.322172682 [1210569-1210604] ../src/vabackend.c:1919 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
    138512.322189164 [1210569-1210604] ../src/vabackend.c: 674           nvCreateConfig got profile: 18 with 0 attributes
    138512.322194052 [1210569-1210604] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 8 (10) (nil) 0
    138512.322196180 [1210569-1210604] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 8 (10) 0x79c6cf116280 8
    138512.328505018 [1210569-1210604] ../src/vabackend.c:1919 nvQuerySurfaceAttributes Returning constraints: width: 144 - 8192, height: 144 - 8192
    138512.328519271 [1210569-1210604] ../src/vabackend.c: 674           nvCreateConfig got profile: 21 with 0 attributes
[... cut ...]
    138540.388262664 [1210569-1211416] ../src/vabackend.c: 327           destroyContext Waiting for resolve thread to exit
    138540.388268289 [1210569-1211422] ../src/vabackend.c: 458          resolveSurfaces [RT] Resolve thread for 0x79c6b8c19000 exiting
    138540.388288502 [1210569-1211416] ../src/vabackend.c: 329           destroyContext pthread_timedjoin_np finished with 0
```

</details>

Archlinux
libva-nvidia-driver-git with https://github.com/elFarto/nvidia-vaapi-driver/pull/366 patch
firefox 138.0.1-1
nvidia-dkms 570.144-3

thesword53, 2025-05-03:
Sorry, it doesn't seem to be the same issue as https://github.com/elFarto/nvidia-vaapi-driver/issues/359 and I can't test the video with my GPU (RTX 2080 SUPER) as it doesn't support AV1 video decoding.

Doaxan, 2025-05-04:
> > [@thesword53](https://github.com/thesword53) unfortunately:
> 
> I modified [#366](https://github.com/elFarto/nvidia-vaapi-driver/pull/366) can you check again if you still have the issue ?

Same here

```
doaxan@b660m ~> NVD_LOG=1 MOZ_GFX_DEBUG=1 firefox-developer-edition
GLX_TEST: childgltest start
GLX_TEST: get_pci_status start
GLX_TEST: get_pci_status finished
GLX_TEST: wayland_egltest start
GLX_TEST: get_egl_status start
GLX_TEST: get_egl_gl_status start
GLX_TEST: get_egl_gl_status finished
GLX_TEST: get_egl_status finished with return: 1
GLX_TEST: wayland_egltest finished
PCI_VENDOR_ID
0x10de
PCI_DEVICE_ID
0x2204
VENDOR
NVIDIA Corporation
RENDERER
NVIDIA GeForce RTX 3090/PCIe/SSE2
VERSION
4.6.0 NVIDIA 570.144
TFP
TRUE
DRM_RENDERDEVICE
/dev/dri/renderD128
MESA_VENDOR_ID
0x10de
MESA_DEVICE_ID
0x2204
TEST_TYPE
EGL
GLX_TEST: childgltest finished
vaapitest start, device /dev/dri/renderD128
vaInitialize finished
Profile: H264Main
Profile: H264High
Profile: H264ConstrainedBaseline
Profile: HEVCMain
Profile: VP8Version0_3
Profile: VP9Profile0
Profile: AV1Profile0
Profile: HEVCMain10
Profile: HEVCMain12
Profile: VP9Profile2
vaapitest finished
VAAPI_SUPPORTED
TRUE
VAAPI_HWCODECS
496
      2636.031204751 [13003-13040] ../src/vabackend.c:2240       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver
      2636.031215742 [13003-13040] ../src/vabackend.c:2248       __vaDriverInit_1_0 Got DRM FD: 1 23
      2636.031219969 [13003-13040] ../src/vabackend.c:2260       __vaDriverInit_1_0 Now have 0 (0 max) instances
      2636.031222799 [13003-13040] ../src/vabackend.c:2287       __vaDriverInit_1_0 Selecting Direct backend
      2636.040205982 [13003-13040] ../src/direct/nv-driver.c: 305            init_nvdriver Initing nvdriver...
      2636.040237101 [13003-13040] ../src/direct/nv-driver.c: 323            init_nvdriver NVIDIA kernel driver version: 570.144, major version: 570, minor version: 144
      2636.040241735 [13003-13040] ../src/direct/nv-driver.c: 330            init_nvdriver Got dev info: 100 1 2 6
      2636.148362212 [13003-13040] ../src/vabackend.c:1486      nvQueryImageFormats In nvQueryImageFormats
      2636.332400448 [13003-13040] ../src/vabackend.c: 674           nvCreateConfig got profile: 6 with 0 attributes
      2636.332418118 [13003-13040] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      2636.332421722 [13003-13040] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 4 (8) 0x7cb9f8c0d940 8
      2636.341648336 [13003-13040] ../src/vabackend.c:1919 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      2636.341667684 [13003-13040] ../src/vabackend.c: 674           nvCreateConfig got profile: 7 with 0 attributes
      2636.341672693 [13003-13040] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      2636.341674508 [13003-13040] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 4 (8) 0x7cb9f8c0d940 8
[... 343 lines cut ...]
      2650.981391018 [13003-14213] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x7cb9f6783300 1920x1080
      2650.990941672 [13003-14213] ../src/vabackend.c: 971        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x7cb9f38f13e0)
      2650.991191954 [13003-14213] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x7cb9f6783640 1920x1080
      2651.007389154 [13003-14213] ../src/vabackend.c: 971        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x7cb9f27ce9c0)
      2651.007566644 [13003-14213] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x7cb9f6783e60 1920x1080
      2651.024129429 [13003-14213] ../src/vabackend.c: 971        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x7cb9f38f3ac0)
      2651.024282777 [13003-14213] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x7cb9f67822c0 1920x1080
      2651.039940034 [13003-14213] ../src/vabackend.c: 971        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x7cb9f38f3480)
      2651.040098801 [13003-14213] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x7cb9f6782ae0 1920x1080
      2651.073946884 [13003-14213] ../src/vabackend.c: 971        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x7cb9f38f4560)
      2651.074039901 [13003-14213] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x7cb9f6782bb0 1920x1080
      2652.084607511 [13003-14213] ../src/vabackend.c: 971        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x7cb9c40d9060)
      2652.084764034 [13003-14213] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x7cb9f6783f30 1920x1080
      2652.977652080 [13003-14213] ../src/vabackend.c: 971        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x7cb9c40d9240)
      2652.977917426 [13003-14213] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x7cb9c40da870 1920x1080
      2653.159825470 [13003-14213] ../src/vabackend.c: 971        nvCreateSurfaces2 Creating surface 1920x1080, format 1 (0x7cb9c40d8480)
      2653.159913340 [13003-14213] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x7cb9c40db160 1920x1080
[... cut ...]
      2656.445018545 [13003-14214] ../src/vabackend.c: 321           destroyContext Signaling resolve thread to exit
      2656.445021789 [13003-14214] ../src/vabackend.c: 327           destroyContext Waiting for resolve thread to exit
      2656.445028324 [13003-14218] ../src/vabackend.c: 458          resolveSurfaces [RT] Resolve thread for 0x7cb9f41be000 exiting
      2656.445047950 [13003-14214] ../src/vabackend.c: 329           destroyContext pthread_timedjoin_np finished with 0
doaxan@b660m ~> 
```
Thanks for trying to fix this!

KevinMX, 2025-05-16:
Using RTX 4080 SUPER and Arch Linux and was testing with the exact same video as the OP.
Encountering `NS_ERROR_DOM_MEDIA_FATAL_ERR` errors as well, logs are mostly the same so I'll not post again here. 
Tried `libva-nvidia-driver-git-0.0.13.r25.g8e6a439-1` and unfortunately 8K AV1 is still not working.

closed 2026-09-26 by elFarto