elFarto/nvidia-vaapi-driver #369: A lot of videos (temporarily) freeze at start while the audio is playing correctly

Swift42, 2025-04-17:
When I scroll through a site with auto-playing AV1 videos (e.g. 9Gag), I have a strange effect:

Around 50% of all videos have a frozen image at the start, while the audio is playing correctly. Then, after 5 to 10 seconds, the video suddenly kicks in and the video plays normally till the end.
And it's not consistent: Sometimes e.g. video A plays correctly and video B has the freeze effect. Then after a page reload video A freezes and video B plays correctly.
When I disable the hardware decoding, all is running normally (no freeze).

I'm using v0.0.13 with Firefox and a RTX 3050 on Linux Mint 22.1

May have nothing to do with it, but Firefox outputs this a lot:

`src/gbm_drv_common.c:130: GBM-DRV error (get_bytes_per_component): Unknown or not supported format: 943212370`

elFarto, 2025-04-19:
Can you check if this happens if you disable hardware acceleration?

elFarto, 2025-04-19:
Can you grab the log with NVD_LOG=1 set?

tedbyron, 2025-04-25:
> Can you grab the log with NVD_LOG=1 set?

I'm getting the same error on youtube, RTX 3080ti, same driver and firefox versions, but I don't have any videos freezing.

<details><summary>Log from hovering a youtube thumbnail</summary>

```console
libva info: VA-API version 1.22.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /run/opengl-driver/lib/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
      2770.105508305 [81062-81097] ../src/vabackend.c:2187       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver: 31
      2770.105511756 [81062-81097] ../src/vabackend.c:2196       __vaDriverInit_1_0 Now have 0 (0 max) instances
      2770.105513645 [81062-81097] ../src/vabackend.c:2222       __vaDriverInit_1_0 Selecting Direct backend
      2770.112046401 [81062-81097] ../src/direct/nv-driver.c: 295            init_nvdriver Initing nvdriver...
      2770.112062985 [81062-81097] ../src/direct/nv-driver.c: 313            init_nvdriver NVIDIA kernel driver version: 565.77, major version: 565, minor version: 77
      2770.112066218 [81062-81097] ../src/direct/nv-driver.c: 320            init_nvdriver Got dev info: 100 1 2 6
libva info: va_openDriver() returns 0
      2770.201355104 [81062-81097] ../src/vabackend.c:1443      nvQueryImageFormats In nvQueryImageFormats
      2770.326328540 [81062-81097] ../src/vabackend.c: 672           nvCreateConfig got profile: 6 with 0 attributes
      2770.326340902 [81062-81097] ../src/vabackend.c:1799 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      2770.326342679 [81062-81097] ../src/vabackend.c:1799 nvQuerySurfaceAttributes with 4 (8) 0x7f2033b31940 8
      2770.336197055 [81062-81097] ../src/vabackend.c:1866 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      2770.336211088 [81062-81097] ../src/vabackend.c: 672           nvCreateConfig got profile: 7 with 0 attributes
      2770.336217031 [81062-81097] ../src/vabackend.c:1799 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      2770.336218762 [81062-81097] ../src/vabackend.c:1799 nvQuerySurfaceAttributes with 4 (8) 0x7f2033b31940 8
      2770.344682032 [81062-81097] ../src/vabackend.c:1866 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      2770.344694701 [81062-81097] ../src/vabackend.c: 672           nvCreateConfig got profile: 13 with 0 attributes
      2770.344698339 [81062-81097] ../src/vabackend.c:1799 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      2770.344699992 [81062-81097] ../src/vabackend.c:1799 nvQuerySurfaceAttributes with 4 (8) 0x7f2033b31940 8
      2770.351067430 [81062-81097] ../src/vabackend.c:1866 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      2770.351076798 [81062-81097] ../src/vabackend.c: 672           nvCreateConfig got profile: 17 with 0 attributes
      2770.351079992 [81062-81097] ../src/vabackend.c:1799 nvQuerySurfaceAttributes with 8 (8) (nil) 0
      2770.351081604 [81062-81097] ../src/vabackend.c:1799 nvQuerySurfaceAttributes with 8 (8) 0x7f2033b31940 8
      2770.359493557 [81062-81097] ../src/vabackend.c:1866 nvQuerySurfaceAttributes Returning constraints: width: 144 - 8192, height: 144 - 8192
      2770.359507782 [81062-81097] ../src/vabackend.c: 672           nvCreateConfig got profile: 14 with 0 attributes
      2770.359512202 [81062-81097] ../src/vabackend.c:1799 nvQuerySurfaceAttributes with 9 (8) (nil) 0
      2770.359513931 [81062-81097] ../src/vabackend.c:1799 nvQuerySurfaceAttributes with 9 (8) 0x7f2033b31940 8
      2770.367523983 [81062-81097] ../src/vabackend.c:1866 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      2770.367536587 [81062-81097] ../src/vabackend.c: 672           nvCreateConfig got profile: 19 with 0 attributes
      2770.367540757 [81062-81097] ../src/vabackend.c:1799 nvQuerySurfaceAttributes with 10 (8) (nil) 0
      2770.367542307 [81062-81097] ../src/vabackend.c:1799 nvQuerySurfaceAttributes with 10 (8) 0x7f2033b31940 8
      2770.377133958 [81062-81097] ../src/vabackend.c:1866 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
      2770.377149675 [81062-81097] ../src/vabackend.c: 672           nvCreateConfig got profile: 32 with 0 attributes
      2770.377152174 [81062-81097] ../src/vabackend.c: 735           nvCreateConfig Unable to determine surface type for VP9/AV1 codec due to no RTFormat specified.
      2770.377156678 [81062-81097] ../src/vabackend.c:1799 nvQuerySurfaceAttributes with 11 (8) (nil) 0
      2770.377158374 [81062-81097] ../src/vabackend.c:1799 nvQuerySurfaceAttributes with 11 (8) 0x7f2033b31940 8
      2770.384459770 [81062-81097] ../src/vabackend.c:1866 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
      2770.384474123 [81062-81097] ../src/vabackend.c: 672           nvCreateConfig got profile: 18 with 0 attributes
      2770.384478825 [81062-81097] ../src/vabackend.c:1799 nvQuerySurfaceAttributes with 8 (10) (nil) 0
      2770.384480516 [81062-81097] ../src/vabackend.c:1799 nvQuerySurfaceAttributes with 8 (10) 0x7f2033b31940 8
      2770.393884739 [81062-81097] ../src/vabackend.c:1866 nvQuerySurfaceAttributes Returning constraints: width: 144 - 8192, height: 144 - 8192
      2770.393900388 [81062-81097] ../src/vabackend.c: 672           nvCreateConfig got profile: 21 with 0 attributes
[... cut ...]
      VAProfileHEVCMain12             : VAEntrypointVLD
      VAProfileVP9Profile2            : VAEntrypointVLD
      VAProfileHEVCMain444            : VAEntrypointVLD
      VAProfileHEVCMain444_10         : VAEntrypointVLD
      VAProfileHEVCMain444_12         : VAEntrypointVLD
      3414.203306120 [100247-100247] ../src/vabackend.c:2095              nvTerminate Terminating 0x37b510c0
      3414.203794488 [100247-100247] ../src/vabackend.c:2109              nvTerminate Now have 0 (0 max) instances
```

</details>

Swift42, 2025-04-25:
Here is the log (while I was scrolling through the site and some videos were delayed):

[nvd_log.txt](https://github.com/user-attachments/files/19906050/nvd_log.txt)

Swift42, 2025-05-10:
I tried to solve it by:
- using an older version (0.0.11, 0.0.12)
- lowering the screen resolution
- using an older nvidia driver (e.g. 535 instead 550)

But the described freezing persists. I don't know what else I can try.

Swift42, 2025-05-15:
I updated the NVidia driver from 550 to 570.
Sadly the problem persists. I'm out of ideas.

Swift42, 2025-06-25:
I think you did it!
With the latest 0.0.14 I have no freezing anymore.
Thank you so much @elFarto

nerijus, 2025-06-25:
You can close the issue then @Swift42 .

closed 2026-09-26 by elFarto