elFarto/nvidia-vaapi-driver #373: YouTube video playback sometimes doesn't start (black screen, FF spinning)

Lastique, 2025-04-21:
Sometimes, starting video playback on YouTube in Firefox 137.0.2 doesn't start. The video area remains black, with endlessly spinning indicator. After a while, there appears a suggestion to reboot the device if the playback doesn't start.

[image]

This doesn't reproduce every time, and in fact, most of the time playback works.

Firefox is started with `LIBVA_DRIVER_NAME=nvidia`, `MOZ_DISABLE_RDD_SANDBOX=1` and `NVD_BACKEND=direct`, and all the `about:config` properties that are mentioned in the README.md are set. `about:support` shows Hardware Decoding for H264, VP9, VP8, AV1 and HEVC as Supported.

Here is the console output of Firefox when this happened, when run with `NVD_LOG=1`:

[firefox-video-doesnt-start2.log](https://github.com/user-attachments/files/19834080/firefox-video-doesnt-start2.log)

nvidia-vaapi-driver is built from commit 0ef97c1849 with https://github.com/elFarto/nvidia-vaapi-driver/pull/371 applied. Nvidia driver 575.51.02, but I've also seen this happen with 570 series.

Lastique, 2025-04-21:
Starting with `NVD_BACKEND=egl` doesn't work. Starting video playback also doesn't work, and I can see this error in the Firefox output:

```
libva info: VA-API version 1.20.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so
    336215.751507996 [599197-599228] ../src/vabackend.c: 166                     init CUDA ERROR 'initialization error' (3)

libva info: Found init function __vaDriverInit_1_0
    336215.751639788 [599197-599228] ../src/vabackend.c:2239       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver
    336215.751652676 [599197-599228] ../src/vabackend.c:2247       __vaDriverInit_1_0 Got DRM FD: 1 25
    336215.751676635 [599197-599228] ../src/vabackend.c:2259       __vaDriverInit_1_0 Now have 0 (0 max) instances
    336215.751691638 [599197-599228] ../src/vabackend.c:2283       __vaDriverInit_1_0 Selecting EGL backend
    336215.797209489 [599197-599228] ../src/export-buf.c: 144       findGPUIndexFromFd Looking for DRM device index: 129
    336215.802983860 [599197-599228] ../src/export-buf.c: 158       findGPUIndexFromFd Found 4 EGL devices
    336215.871280068 [599197-599228] ../src/export-buf.c: 193       findGPUIndexFromFd No EGL_CUDA_DEVICE_NV support for EGLDevice 0
    336215.871302025 [599197-599228] ../src/export-buf.c: 193       findGPUIndexFromFd No EGL_CUDA_DEVICE_NV support for EGLDevice 1
    336215.871305436 [599197-599228] ../src/export-buf.c: 193       findGPUIndexFromFd No EGL_CUDA_DEVICE_NV support for EGLDevice 2
    336215.871308320 [599197-599228] ../src/export-buf.c: 196       findGPUIndexFromFd No DRM device file for EGLDevice 3
    336215.871309937 [599197-599228] ../src/export-buf.c: 199       findGPUIndexFromFd No match found, falling back to default device
    336215.871312996 [599197-599228] ../src/vabackend.c:2311       __vaDriverInit_1_0 Exporter failed
libva error: /usr/lib/x86_64-linux-gnu/dri/nvidia_drv_video.so init failed
libva info: va_openDriver() returns 1
```

elFarto, 2025-06-22:
I'm not sure what's going on here. Does this still happen with the latest release of this driver?