elFarto/nvidia-vaapi-driver #375: HDR Support

jp7677, 2025-05-10:
Hi there,

Thanks a lot for your great project! Running Fedora 42 Gnome Desktop, viewing HDR content using an AMD card is nowadays relatively straight foward. HDR needs to be toggled on in the Gnome settings and with firefox 137 the following two settings needs to be switch to `true` for allowing e.g. Youtube HDR:

- `gfx.wayland.hdr`
- `layout.css.video-dynamic-range.allows-high`

Unfortunately playback crashes firefox on an NVIDIA  card with the `nvidia-vaapi-driver`: 

<details><summary>firefox logging with 8e6a439600137d9105d238b810857e185cfe4cec</summary>
<p>

```
1933.690234322 [12271-12309] ../src/vabackend.c:2240       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver (Development-Snapshot)
      1933.690244733 [12271-12309] ../src/vabackend.c:2248       __vaDriverInit_1_0 Got DRM FD: 1 23
      1933.690250106 [12271-12309] ../src/vabackend.c:2260       __vaDriverInit_1_0 Now have 0 (0 max) instances
      1933.690255596 [12271-12309] ../src/vabackend.c:2287       __vaDriverInit_1_0 Selecting Direct backend
      1933.711714230 [12271-12309] ../src/direct/nv-driver.c: 305            init_nvdriver Initing nvdriver...
      1933.711739952 [12271-12309] ../src/direct/nv-driver.c: 323            init_nvdriver NVIDIA kernel driver version: 570.144, major version: 570, minor version: 144
      1933.711743475 [12271-12309] ../src/direct/nv-driver.c: 330            init_nvdriver Got dev info: 100 1 2 6
      1933.806983357 [12271-12309] ../src/vabackend.c:1486      nvQueryImageFormats In nvQueryImageFormats
      1933.973506102 [12271-12309] ../src/vabackend.c: 674           nvCreateConfig got profile: 6 with 0 attributes
      1933.973521041 [12271-12309] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      1933.973523401 [12271-12309] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 4 (8) 0x7f4608b2edc0 8
      1933.982749435 [12271-12309] ../src/vabackend.c:1919 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      1933.982768980 [12271-12309] ../src/vabackend.c: 674           nvCreateConfig got profile: 7 with 0 attributes
      1933.982774008 [12271-12309] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      1933.982775981 [12271-12309] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 4 (8) 0x7f4608b2edc0 8
      1933.991110865 [12271-12309] ../src/vabackend.c:1919 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      1933.991129765 [12271-12309] ../src/vabackend.c: 674           nvCreateConfig got profile: 13 with 0 attributes
      1933.991134390 [12271-12309] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 4 (8) (nil) 0
      1933.991136273 [12271-12309] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 4 (8) 0x7f4608b2edc0 8
      1933.998948331 [12271-12309] ../src/vabackend.c:1919 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      1933.998968517 [12271-12309] ../src/vabackend.c: 674           nvCreateConfig got profile: 17 with 0 attributes
      1933.998973645 [12271-12309] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 8 (8) (nil) 0
      1933.998975486 [12271-12309] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 8 (8) 0x7f4608b2ee80 8
      1934.007433824 [12271-12309] ../src/vabackend.c:1919 nvQuerySurfaceAttributes Returning constraints: width: 144 - 8192, height: 144 - 8192
      1934.007460452 [12271-12309] ../src/vabackend.c: 674           nvCreateConfig got profile: 14 with 0 attributes
      1934.007466263 [12271-12309] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 9 (8) (nil) 0
      1934.007468319 [12271-12309] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 9 (8) 0x7f4608b2edc0 8
      1934.016007833 [12271-12309] ../src/vabackend.c:1919 nvQuerySurfaceAttributes Returning constraints: width: 48 - 4096, height: 16 - 4096
      1934.016029242 [12271-12309] ../src/vabackend.c: 674           nvCreateConfig got profile: 19 with 0 attributes
      1934.016033962 [12271-12309] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 10 (8) (nil) 0
      1934.016035780 [12271-12309] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 10 (8) 0x7f4608b2edc0 8
      1934.023817778 [12271-12309] ../src/vabackend.c:1919 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
      1934.023835476 [12271-12309] ../src/vabackend.c: 674           nvCreateConfig got profile: 32 with 0 attributes
      1934.023838795 [12271-12309] ../src/vabackend.c: 737           nvCreateConfig Unable to determine surface type for VP9/AV1 codec due to no RTFormat specified.
      1934.023843479 [12271-12309] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 11 (8) (nil) 0
      1934.023846005 [12271-12309] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 11 (8) 0x7f4608b2edc0 8
      1934.031778525 [12271-12309] ../src/vabackend.c:1919 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
      1934.031803215 [12271-12309] ../src/vabackend.c: 674           nvCreateConfig got profile: 18 with 0 attributes
      1934.031810044 [12271-12309] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 8 (10) (nil) 0
      1934.031813352 [12271-12309] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 8 (10) 0x7f4608b2edc0 8
      1934.040276077 [12271-12309] ../src/vabackend.c:1919 nvQuerySurfaceAttributes Returning constraints: width: 144 - 8192, height: 144 - 8192
      1934.040298464 [12271-12309] ../src/vabackend.c: 674           nvCreateConfig got profile: 21 with 0 attributes
      1934.040304201 [12271-12309] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 10 (10) (nil) 0
      1934.040306035 [12271-12309] ../src/vabackend.c:1852 nvQuerySurfaceAttributes with 10 (10) 0x7f4608b2edc0 8
      1934.048246722 [12271-12309] ../src/vabackend.c:1919 nvQuerySurfaceAttributes Returning constraints: width: 128 - 8192, height: 128 - 8192
      1934.048598950 [12271-12599] ../src/vabackend.c:2148              nvTerminate Terminating 0x7f46317f4300
      1934.049237533 [12271-12599] ../src/vabackend.c:2162              nvTerminate Now have 0 (0 max) instances
      1934.108822390 [12271-12309] ../src/vabackend.c:2240       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver (Development-Snapshot)
      1934.108832695 [12271-12309] ../src/vabackend.c:2248       __vaDriverInit_1_0 Got DRM FD: 1 23
      1934.108835730 [12271-12309] ../src/vabackend.c:2260       __vaDriverInit_1_0 Now have 0 (0 max) instances
      1934.108841342 [12271-12309] ../src/vabackend.c:2287       __vaDriverInit_1_0 Selecting Direct backend
      1934.108849864 [12271-12309] ../src/direct/nv-driver.c: 305            init_nvdriver Initing nvdriver...
      1934.108860771 [12271-12309] ../src/direct/nv-driver.c: 323            init_nvdriver NVIDIA kernel driver version: 570.144, major version: 570, minor version: 144
      1934.108863921 [12271-12309] ../src/direct/nv-driver.c: 330            init_nvdriver Got dev info: 100 1 2 6
      1934.210448399 [12271-12309] ../src/vabackend.c:1486      nvQueryImageFormats In nvQueryImageFormats
      1934.210538979 [12271-12309] ../src/vabackend.c:1486      nvQueryImageFormats In nvQueryImageFormats
      1934.244103038 [12271-12309] ../src/vabackend.c:1486      nvQueryImageFormats In nvQueryImageFormats
[... 464 lines cut ...]
      2867.507644411 [13767-13780] ../src/vabackend.c:1006        nvDestroySurfaces Destroying surface -1 (0x7f0ef54d3f70)
      2867.507646157 [13767-13780] ../src/vabackend.c:1006        nvDestroySurfaces Destroying surface -1 (0x7f0ef4e803c0)
[vo/dmabuf-wayland/vaapi] Done probing surface formats.
      2867.507654212 [13767-13780] ../src/vabackend.c:2148              nvTerminate Terminating 0x7f0ef40116b0
      2867.508144235 [13767-13780] ../src/vabackend.c:2162              nvTerminate Now have 0 (0 max) instances
[vo/dmabuf-wayland] Loading failed.
[vo/dmabuf-wayland] Loading hwdec driver 'cuda'
[vo/dmabuf-wayland/cuda] CUDA hwdec only works with OpenGL or Vulkan backends.
[vo/dmabuf-wayland] Loading failed.
[vo/dmabuf-wayland] Loading hwdec driver 'vdpau-gl'
[vo/dmabuf-wayland] Loading failed.
[vo/dmabuf-wayland] Loading hwdec driver 'drmprime'
[vo/dmabuf-wayland/drmprime] Using DRM device: /dev/dri/renderD128
[vo/dmabuf-wayland] Loading hwdec driver 'drmprime-overlay'
[vo/dmabuf-wayland/drmprime-overlay] Failed to retrieve DRM fd from native display.
[vo/dmabuf-wayland] Loading failed.
[vo/dmabuf-wayland] Loading hwdec driver 'vulkan'
[vo/dmabuf-wayland/vulkan] This is not a libplacebo vulkan gpu api context.
[vo/dmabuf-wayland] Assuming 119.990000 FPS for display sync.
``` 

</p>
</details> 

I'm not familiar with the video stack, let alone HDR. Is HDR playback with the `nvidia-vaapi-driver` feasible at this time? Does this needs more work elsewhere in the stack (NVIDIA driver? Gnome? Firefox?) first? Thanks!

jp7677, 2025-05-25:
Related: https://forums.developer.nvidia.com/t/vulkan-extensions-needed-for-hdr-is-missing/334268

elFarto, 2025-06-22:
In theory everything should already work. There's nothing particularly special about HDR from the decoders point of view, only that we decode to P010 buffers rather than NV12 ones. It's possible I'm not setting something correctly, but there's not much to set.

There is [this issue](https://github.com/NVIDIA/egl-wayland/issues/41) that appears to be related.

jp7677, 2025-06-29:
> In theory everything should already work. There's nothing particularly special about HDR from the decoders point of view, only that we decode to P010 buffers rather than NV12 ones. It's possible I'm not setting something correctly, but there's not much to set.

Nice to read that there isn't any major work ahead at this front here. 
 
> There is [this issue](https://github.com/NVIDIA/egl-wayland/issues/41) that appears to be related.

Thanks for the link. Reading that issue, that seems to be relevant for prime setups. I'm running just a simple desktop with a single Ampere GPU.

closed 2026-09-26 by elFarto