elFarto/nvidia-vaapi-driver #379: Firefox Crashes on Fullscreen Video with NVIDIA Hardware Acceleration on Wayland

SamuelBorn, 2025-05-27:
**Description:**
Firefox crashes when attempting to fullscreen a video (e.g., YouTube) after hardware video acceleration (VA-API) has been successfully enabled and confirmed working via `nvidia-smi dmon` (decoder usage increases). The crash occurs specifically only when entering fullscreen mode.

**System Information:**
* **GPU:** NVIDIA GeForce GTX 1650 Mobile / Max-Q
* **OS:** Fedora Linux 42 (using Wayland)
* **Browser:** Mozilla Firefox 138.0.4

**Steps to Reproduce:**
1.  Enable hardware video acceleration in Firefox.
2.  Open a YouTube video.
3.  Observe `nvidia-smi dmon` showing decoder activity (e.g., `dec` column increases).
4.  Attempt to switch the video to fullscreen mode.

**Expected Behavior:**
The video plays in fullscreen mode without Firefox crashing.

**Actual Behavior:**
Firefox crashes immediately upon entering fullscreen. The following error is observed in the terminal:

```
libva info: VA-API version 1.22.0
libva info: User environment variable requested driver 'nvidia'
libva info: Trying to open /usr/lib64/dri-nonfree/nvidia_drv_video.so
libva info: Trying to open /usr/lib64/dri-freeworld/nvidia_drv_video.so
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
libva info: va_openDriver() returns 0
[GFX1-]: (gnome) Wayland protocol error: [destroyed object]: error 7: failed to import supplied dmabufs: Could not bind the given EGLImage to a CoglTexture2D

ExceptionHandler::GenerateDump attempting to generate:/home/born/.mozilla/firefox/0f4adum9.default-release/minidumps/698a952d-bc27-712f-182c-14023ccab212.dmp
ExceptionHandler::GenerateDump cloned child 77761
ExceptionHandler::SendContinueSignalToChild sent continue signal to child
ExceptionHandler::WaitForContinueSignal waiting for continue signal...
ExceptionHandler::GenerateDump minidump generation succeeded
Exiting due to channel error.
```

**Key Error Message:**
`[GFX1-]: (gnome) Wayland protocol error: [destroyed object]: error 7: failed to import supplied dmabufs: Could not bind the given EGLImage to a CoglTexture2D`

elFarto, 2025-06-22:
I've just reconfirmed this still works on my machine, and it does. Can you confirm which GPU your wayland compositor is using? I'm assuming it's using built-in one, which would cause the failure when attempting to import an exported DMA-BUF.

I'm not sure we'll ever get cross-GPU export/imports working correctly.

SamuelBorn, 2025-07-01:
Actually I am not getting firefox to work at all at the moment...
I would love to debug this further but I have no idea what is wrong with my current configuration.
Maybe you have any Idea what is currently wrong with my setup.

---

- Downloaded `libva-nvidia-driver`
- Add `nvidia-drm.modeset=1` to kernel params
- Set the following Env vars:
```
export MOZ_DISABLE_RDD_SANDBOX=1
export NVD_BACKEND=direct
export LIBVA_DRIVER_NAME=nvidia
```
- Set all of those firefox options
```
media.ffmpeg.vaapi.enabled = true
media.hardware-video-decoding.force-enabled = true
media.rdd-ffmpeg.enabled = true
media.av1.enabled = false
gfx.x11-egl.force-enabled = true
widget.dmabuf.force-enabled = true
```
- Installed `enhanced-h264ify` firefox extension

`WebRenderer` enabled
[image]

`HARDWARE_VIDEO_DECODING` force enabled
[image]

Codec seem supported
[image]

MPV hardware acceleration via `mpv --hwdec=auto ...` works
[image]

thesword53, 2025-07-27:
> Firefox + Youtube still uses the Intel GPU 😦

You should run the whole Firefox with the Nvidia GPU (`prime-run firefox`) because if you run Firefox with the Intel GPU and it decodes videos with the Nvidia GPU, it will fail when importing dma-bufs from Nvidia GPU to Intel GPU as you show in logs because of Nvidia driver limitation. In your case I suggest you use the Intel GPU for decoding as it supports the same video codecs as the Nvidia and it will draw less power (https://forums.developer.nvidia.com/t/nvdec-forces-gpu-into-p2-cuda-state-much-higher-power-consumption-than-with-vdpau/55466).

SamuelBorn, 2025-07-30:
> prime-run firefox

That worked! Now my graphic card does something when watching videos. Thank you ❤️ 

To make this change persistent would you recommend adding it to `exec` line of the `firefox.desktop` file?

Obdam, 2025-08-27:
FWIW: removing the libva_driver_name environment variable + running Firefox with `prime-run firefox` also worked using Hyprland on Arch Linux.

danilw, 2026-01-05:
I have not read full discussion - only answer to header

> Firefox Crashes on Fullscreen Video with NVIDIA Hardware Acceleration on Wayland

im on Firefox 144.0.2 
Nvidia driver 580.105.08 (monitor connected to GPU - iGPU not used)
Wayland-Gnome - no additional firefox options than in this setup how-to

decoding confirmed to working
**sreenshot 1080p full screen - no crashes**

[image]

closed 2026-09-26 by elFarto