elFarto/nvidia-vaapi-driver #382: Ubuntu 25.04: Dependency "libdrm" not found

tredondo, 2025-06-11:
```
$ meson setup build
The Meson build system
Version: 1.7.0
Source dir: ./nvidia-vaapi-driver-0.0.14
Build dir: ./nvidia-vaapi-driver-0.0.14/build
Build type: native build
Project name: nvidia-vaapi-driver
Project version: 0.1
C compiler for the host machine: cc (gcc 14.2.0 "cc (Ubuntu 14.2.0-19ubuntu2) 14.2.0")
C linker for the host machine: cc ld.bfd 2.44
Host machine cpu family: x86_64
Host machine cpu: x86_64
Library m found: YES
Library dl found: YES
Found pkg-config: YES (/usr/bin/pkg-config) 1.8.1
Run-time dependency egl found: YES 1.5
Run-time dependency ffnvcodec found: YES 12.1.14.0
Found CMake: /usr/bin/cmake (3.31.6)
Run-time dependency libdrm found: NO (tried pkgconfig and cmake)

meson.build:21:4: ERROR: Dependency "libdrm" not found, tried pkgconfig and cmake
```

I tried to `sudo apt install libdrm` but got "Error: Unable to locate package libdrm".

thesword53, 2025-06-19:
> I tried to sudo apt install libdrm but got "Error: Unable to locate package libdrm".
You also need the developer package `sudo apt install libdrm-dev`

elFarto, 2025-06-22:
I'll add that library to the list in the readme.

closed 2025-08-07 by elFarto