elFarto/nvidia-vaapi-driver #386: Video corruption when using chromium

ryao, 2025-08-02:
The following URL has video corruption when using chromium version 137.0.7151.103 with nvidia driver version 575.64:

https://store.steampowered.com/app/222880/Insurgency/

I used obtained a log using NVD_LOG=1:

```
$ env NVD_LOG=1 chromium --enable-features=AcceleratedVideoDecodeLinuxZeroCopyGL,AcceleratedVideoDecodeLinuxGL,VaapiIgnoreDriverChecks,VaapiOnNvidiaGPUs,AcceleratedVideoEncoder
    938060.085287487 [4007-4007] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:2260       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver
    938060.085301403 [4007-4007] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:2268       __vaDriverInit_1_0 Got DRM FD: 1 44
    938060.085309298 [4007-4007] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:2280       __vaDriverInit_1_0 Now have 0 (0 max) instances
    938060.085313456 [4007-4007] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:2307       __vaDriverInit_1_0 Selecting Direct backend
    938060.085328815 [4007-4007] ../nvidia-vaapi-driver-0.0.14/src/direct/nv-driver.c: 305            init_nvdriver Initing nvdriver...
    938060.085347310 [4007-4007] ../nvidia-vaapi-driver-0.0.14/src/direct/nv-driver.c: 323            init_nvdriver NVIDIA kernel driver version: 575.64, major version: 575, minor version: 64
    938060.085352630 [4007-4007] ../nvidia-vaapi-driver-0.0.14/src/direct/nv-driver.c: 330            init_nvdriver Got dev info: 800 1 2 6
    938060.590328759 [4007-4007] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c: 754           nvCreateConfig Unable to determine surface type for VP9/AV1 codec due to no RTFormat specified.
    938060.613161113 [4007-4007] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:2168              nvTerminate Terminating 0x12dc03093400
    938060.614573677 [4007-4007] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:2182              nvTerminate Now have 0 (0 max) instances
    938066.499642382 [4007-4649] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:2260       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver
    938066.499656488 [4007-4649] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:2268       __vaDriverInit_1_0 Got DRM FD: 1 44
    938066.499663552 [4007-4649] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:2280       __vaDriverInit_1_0 Now have 0 (0 max) instances
    938066.499670876 [4007-4649] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:2307       __vaDriverInit_1_0 Selecting Direct backend
    938066.499705220 [4007-4649] ../nvidia-vaapi-driver-0.0.14/src/direct/nv-driver.c: 305            init_nvdriver Initing nvdriver...
    938066.499734365 [4007-4649] ../nvidia-vaapi-driver-0.0.14/src/direct/nv-driver.c: 323            init_nvdriver NVIDIA kernel driver version: 575.64, major version: 575, minor version: 64
    938066.499739936 [4007-4649] ../nvidia-vaapi-driver-0.0.14/src/direct/nv-driver.c: 330            init_nvdriver Got dev info: 800 1 2 6
    938066.865895948 [4007-4649] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:1051          nvCreateContext Creating context with 0 render targets, at 854x480
    938066.914776471 [4007-4649] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:1134          nvCreateContext Creating decoder: 0x12dc08030000 for context id: 2
    938066.914894073 [4007-4649] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c: 988        nvCreateSurfaces2 Creating surface 854x480, format 1 (0x12dc042e7540)
    938066.914902238 [4007-4660] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c: 403          resolveSurfaces [RT] Resolve thread for 0x12dc047c9000 started
    938066.914905344 [4007-4649] ../nvidia-vaapi-driver-0.0.14/src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x12dc05960540 854x480
    938066.915887136 [4007-4649] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c: 988        nvCreateSurfaces2 Creating surface 854x480, format 1 (0x12dc04574480)
    938066.915911161 [4007-4649] ../nvidia-vaapi-driver-0.0.14/src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x12dc03c379c0 854x480
    938066.916944991 [4007-4649] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:1372          nvRenderPicture Unhandled buffer type: 1
    938066.916954559 [4007-4649] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:1372          nvRenderPicture Unhandled buffer type: 13
<snip>
```

According to chromium's dev tools, the video is using VP8.

When I launch firefox (139.0.4), there is no corruption:

```
$ env MOZ_DISABLE_RDD_SANDBOX=1 NVD_LOG=1 firefox
libva info: VA-API version 1.22.0
libva info: Trying to open /usr/lib64/va/drivers/nvidia_drv_video.so
libva info: Found init function __vaDriverInit_1_0
    937926.727742920 [2955-2988] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:2260       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver
    937926.727750915 [2955-2988] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:2268       __vaDriverInit_1_0 Got DRM FD: 1 22
    937926.727769290 [2955-2988] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:2280       __vaDriverInit_1_0 Now have 0 (0 max) instances
    937926.727775892 [2955-2988] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:2307       __vaDriverInit_1_0 Selecting Direct backend
    937926.762418631 [2955-2988] ../nvidia-vaapi-driver-0.0.14/src/direct/nv-driver.c: 305            init_nvdriver Initing nvdriver...
    937926.762459599 [2955-2988] ../nvidia-vaapi-driver-0.0.14/src/direct/nv-driver.c: 323            init_nvdriver NVIDIA kernel driver version: 575.64, major version: 575, minor version: 64
    937926.762467494 [2955-2988] ../nvidia-vaapi-driver-0.0.14/src/direct/nv-driver.c: 330            init_nvdriver Got dev info: 800 1 2 6
libva info: va_openDriver() returns 0
    937927.225845245 [2955-2988] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c: 754           nvCreateConfig Unable to determine surface type for VP9/AV1 codec due to no RTFormat specified.
    937927.290195585 [2955-3439] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c: 988        nvCreateSurfaces2 Creating surface 854x480, format 1 (0x7fb6bb696560)
    937927.305817530 [2955-3439] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:1051          nvCreateContext Creating context with 0 render targets, at 854x480
    937927.352884742 [2955-3439] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:1134          nvCreateContext Creating decoder: 0x7fb6bb72d000 for context id: 13
    937927.352975213 [2955-3442] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c: 403          resolveSurfaces [RT] Resolve thread for 0x7fb6ba13c000 started
    937927.353002494 [2955-3439] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:1372          nvRenderPicture Unhandled buffer type: 13
    937927.353012373 [2955-3439] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:1372          nvRenderPicture Unhandled buffer type: 1
    937927.353050405 [2955-3439] ../nvidia-vaapi-driver-0.0.14/src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x7fb6be66ee60 854x480
    937927.457771478 [2955-3439] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:1170         nvDestroyContext Destroying context: 13
    937927.457796064 [2955-3439] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c: 321           destroyContext Signaling resolve thread to exit
    937927.457807546 [2955-3439] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c: 327           destroyContext Waiting for resolve thread to exit
    937927.584663137 [2955-3442] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c: 458          resolveSurfaces [RT] Resolve thread for 0x7fb6ba13c000 exiting
    937927.584694596 [2955-3439] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c: 329           destroyContext Finished waiting for resolve thread with 0
    937927.584712901 [2955-3439] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:1023        nvDestroySurfaces Destroying surface 0 (0x7fb6bb696560)
    937927.584925161 [2955-3439] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:2168              nvTerminate Terminating 0x7fb8e5f6eb60
    937927.586149791 [2955-3439] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:2182              nvTerminate Now have 0 (0 max) instances
[... 110 lines cut ...]
    938697.276485212 [16199-16965] ../nvidia-vaapi-driver-0.0.14/src/export-buf.c:  97                reconnect Reconnecting to stream
    938697.276630958 [16199-16965] ../nvidia-vaapi-driver-0.0.14/src/export-buf.c: 117                reconnect CUDA ERROR 'operation not supported' (801)

[... cut ...]
    938697.508708223 [16199-16965] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:2168              nvTerminate Terminating 0x7f9fadb6d4e0
    938697.508713884 [16199-16965] ../nvidia-vaapi-driver-0.0.14/src/export-buf.c:  63      egl_releaseExporter Releasing exporter, 0 outstanding frames
    938697.509797263 [16199-16965] ../nvidia-vaapi-driver-0.0.14/src/export-buf.c:  80      egl_releaseExporter Done releasing frames
    938697.509838501 [16199-16965] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:2182              nvTerminate Now have 0 (0 max) instances
```

fravchina, 2025-08-04:
```
    938066.916944991 [4007-4649] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:1372          nvRenderPicture Unhandled buffer type: 1
    938066.916954559 [4007-4649] ../nvidia-vaapi-driver-0.0.14/src/vabackend.c:1372          nvRenderPicture Unhandled buffer type: 13
```

Those are 1=VAIQMatrixBufferType and 13=VAProbabilityBufferType. They are not implemented for vp8: https://github.com/elFarto/nvidia-vaapi-driver/blob/677f48002cee82e4e37d4e95a5b085ab1c5bbe98/src/vp8.c#L66

Seems like it's required to implement them for vp8 in chromium:
https://source.chromium.org/chromium/chromium/src/+/main:media/gpu/vaapi/vaapi_webp_decoder.cc;drc=f5d8b7440713112f54f04fd94180897c76497422;l=160
https://source.chromium.org/chromium/chromium/src/+/main:media/gpu/vaapi/vp8_vaapi_video_decoder_delegate.cc;drc=37a540f8541c23040242a3de6c68d69618ced167;l=63

elFarto, 2025-08-07:
The video on the Insurgency page is VP8, not VP9. My card doesn't support VP8 so I can't reproduce this issue.

ryao, 2025-08-08:
> The video on the Insurgency page is VP8, not VP9. My card doesn't support VP8 so I can't reproduce this issue.

@elFarto Which card are you using?

Sid127, 2025-08-10:
@ryao I'd love it if you could cross verify my PR to see if it fixes chromium playback, at least :)

VeilSilence, 2025-08-10:
I'm afraid your patch didn't fix issue for me at least. Driver version 580 "Open" .
<details><summary>Screen again</summary>
<p>
[image]
</p>
</details> 

[Brave:gpu_log.txt](https://github.com/user-attachments/files/21705276/Brave.gpu_log.txt)
<details><summary>NVD_LOG</summary>
<p>

```
       298.487566984 [6334-6334] ../src/vabackend.c:2260       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver
       298.487576414 [6334-6334] ../src/vabackend.c:2268       __vaDriverInit_1_0 Got DRM FD: 1 46
       298.487582344 [6334-6334] ../src/vabackend.c:2280       __vaDriverInit_1_0 Now have 0 (0 max) instances
       298.487588644 [6334-6334] ../src/vabackend.c:2307       __vaDriverInit_1_0 Selecting Direct backend
       298.487602814 [6334-6334] ../src/direct/nv-driver.c: 305            init_nvdriver Initing nvdriver...
       298.487617054 [6334-6334] ../src/direct/nv-driver.c: 323            init_nvdriver NVIDIA kernel driver version: 580.65.06, major version: 580, minor version: 65
       298.487620434 [6334-6334] ../src/direct/nv-driver.c: 330            init_nvdriver Got dev info: b00 1 2 6
       298.859833954 [6334-6334] ../src/vabackend.c: 754           nvCreateConfig Unable to determine surface type for VP9/AV1 codec due to no RTFormat specified.
       298.876627476 [6334-6334] ../src/vabackend.c:2168              nvTerminate Terminating 0x2fec00a23e00
       298.877342017 [6334-6334] ../src/vabackend.c:2182              nvTerminate Now have 0 (0 max) instances
       307.203660942 [6334-6931] ../src/vabackend.c:2260       __vaDriverInit_1_0 Initialising NVIDIA VA-API Driver
       307.203669372 [6334-6931] ../src/vabackend.c:2268       __vaDriverInit_1_0 Got DRM FD: 1 46
       307.203675782 [6334-6931] ../src/vabackend.c:2280       __vaDriverInit_1_0 Now have 0 (0 max) instances
       307.203679102 [6334-6931] ../src/vabackend.c:2307       __vaDriverInit_1_0 Selecting Direct backend
       307.203706422 [6334-6931] ../src/direct/nv-driver.c: 305            init_nvdriver Initing nvdriver...
       307.203728402 [6334-6931] ../src/direct/nv-driver.c: 323            init_nvdriver NVIDIA kernel driver version: 580.65.06, major version: 580, minor version: 65
       307.203734372 [6334-6931] ../src/direct/nv-driver.c: 330            init_nvdriver Got dev info: b00 1 2 6
       307.498820976 [6334-6931] ../src/vabackend.c:1051          nvCreateContext Creating context with 0 render targets, at 854x480
       307.516444109 [6334-6931] ../src/vabackend.c:1134          nvCreateContext Creating decoder: 0x2fec04e1c000 for context id: 2
       307.516527419 [6334-6931] ../src/vabackend.c: 988        nvCreateSurfaces2 Creating surface 854x480, format 1 (0x2fec04aedef0)
       307.516533369 [6334-6931] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x2fec02725dc0 854x480
       307.516539099 [6334-6937] ../src/vabackend.c: 403          resolveSurfaces [RT] Resolve thread for 0x2fec04e8c000 started
       307.516943439 [6334-6931] ../src/vabackend.c: 988        nvCreateSurfaces2 Creating surface 854x480, format 1 (0x2fec04aee680)
       307.516949969 [6334-6931] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x2fec02727c60 854x480
       307.517385600 [6334-6931] ../src/vabackend.c: 988        nvCreateSurfaces2 Creating surface 854x480, format 1 (0x2fec04aed550)
       307.517391170 [6334-6931] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x2fec02724540 854x480
       307.517792570 [6334-6931] ../src/vabackend.c: 988        nvCreateSurfaces2 Creating surface 854x480, format 1 (0x2fec04aed810)
       307.517798120 [6334-6931] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x2fec02727480 854x480
       307.518185970 [6334-6931] ../src/vabackend.c: 988        nvCreateSurfaces2 Creating surface 854x480, format 1 (0x2fec04aedfa0)
       307.518191530 [6334-6931] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x2fec027273a0 854x480
       307.518556821 [6334-6931] ../src/vabackend.c: 988        nvCreateSurfaces2 Creating surface 854x480, format 1 (0x2fec04aee940)
       307.518563271 [6334-6931] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x2fec02727020 854x480
       307.519016941 [6334-6931] ../src/vabackend.c: 988        nvCreateSurfaces2 Creating surface 854x480, format 1 (0x2fec04aee050)
       307.519022931 [6334-6931] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x2fec02727100 854x480
       307.519431471 [6334-6931] ../src/vabackend.c: 988        nvCreateSurfaces2 Creating surface 854x480, format 1 (0x2fec04aed130)
       307.519437121 [6334-6931] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x2fec02724700 854x480
       307.519941352 [6334-6931] ../src/vabackend.c: 988        nvCreateSurfaces2 Creating surface 854x480, format 1 (0x2fec04aec8f0)
       307.519949042 [6334-6931] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x2fec02726f40 854x480
       307.520448382 [6334-6931] ../src/vabackend.c: 988        nvCreateSurfaces2 Creating surface 854x480, format 1 (0x2fec04aecbb0)
[... cut ...]
       307.549416473 [6334-6931] ../src/vabackend.c: 988        nvCreateSurfaces2 Creating surface 854x480, format 1 (0x2fec009a6680)
       307.549423513 [6334-6931] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x2fec02727b80 854x480
       307.549881153 [6334-6931] ../src/vabackend.c: 988        nvCreateSurfaces2 Creating surface 854x480, format 1 (0x2fec02897230)
       307.549887603 [6334-6931] ../src/direct/direct-export-buf.c: 190 direct_allocateBackingImage Allocating BackingImages: 0x2fec02727800 854x480
```

</p>
</details>

gre4kapi, 2025-08-17:
Hello, i'm launching discord with these parameters to get HEVC decoding working:
```
--start-minimized --enable-features=AcceleratedVideoDecodeLinuxGL,AcceleratedVideoDecodeLinuxZeroCopyGL,VaapiOnNvidiaGPUs,VaapiIgnoreDriverChecks,UseOzonePlatform,WaylandWindowDecorations --ozone-platform=wayland
```
and it actually works, but i'm getting gif corruption in discord's gif selection, also discord reboots sometimes. You can see both on the video:

https://github.com/user-attachments/assets/0fb7b3c8-bfcc-4190-b06b-26c8d7b36cc1

Any way to fix this?

closed 2025-09-27 by elFarto