elFarto/nvidia-vaapi-driver #391: FEATURE_HARDWARE_VIDEO_DECODING_NO_LINUX_NVIDIA error in Firefox

NielsGx, 2025-08-24:
Hi,

I am running Hyprland on CachyOS, with a RTX 4080.
I have nvidia-open-dkms and libva (this) installed

my environment variables in hyprland:
```
env = LIBVA_DRIVER_NAME,nvidia
env = __GLX_VENDOR_LIBRARY_NAME,nvidia
env = NVD_BACKEND,direct
env = MOZ_ENABLE_WAYLAND,1
env = MOZ_DISABLE_RDD_SANDBOX,1
```

I have also enabled all the things in `about:config` as specified in your readme

But I get this in `about:support`
[image]

(btw it seems to work in Chromium when it shouldn't ?!)
[image]

NielsGx, 2025-09-09:
> Something that is helpful for me when debugging the drivers is running the following command in the terminal while playing. a video:
> 
> `watch -n 2 nvidia-smi `
> 
> This lets you see all the processes on your computer running your graphics card, you should be able to see a noticeable uptick when playing a video in firefox vs. idle firefox, If there is no uptick then its not working.

Apparently the nvidia-open-dkms driver doesn't let you see "Decode" usage in nvidia-smi, so hard to know...

But as I said earlier, the log I got with "VA-API frame" is probably showing it works

NielsGx, 2025-09-25:
> This blocklist reason is perfectly normal and it's just the new name for the old `FEATURE_FAILURE_VAAPI_NO_LINUX_NVIDIA` that existed ever since firefox added hardware acceleration. I guess that back in 2022 it may have just existed for peace of mind to fence the old VDPAU-VAAPI wrapper.
> 
> I could very easily bypass it with `media.hardware-video-decoding.force-enabled`=true.

Obviously, I enabled all the stuff said in the readme..

mirh, 2025-09-26:
Then you should have no problems? The codec support table should give you the first five rows green.

(I will admit that funnily enough checking this again made me realize I was still missing `MOZ_DISABLE_RDD_SANDBOX=1`, but after that 4K VP9 cpu usage went from ~25% down to less than 10%).